Ultimaker/Cura #16477: Cura 5.4 - Memory use crash

GrahamDyer, 2023-08-14:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 5

### Reproduction steps

I'm using Win 11 on Ryzen 7 PC with 32GB of RAM printing via USB on an Ender 5. 

When my version of Cura starts it uses about 140MB of RAM but this climbs quickly getting to 10GB in 2min 43 secs. It reaches 15GB in about 4m30s.

Eventually at about 28GB Firefox tabs start to crash, then the screens go black and about 30 seconds later the print stops as Cura crashes. Forced reboot is the next only option.

### Actual results

Cura and PC crashes

### Expected results

No crash, normal memory usage

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/Ultimaker/Cura/files/12338109/cura.zip)
[image]

ryhola, 2023-08-14:
I have the same problem. I have very similar specs I'm using Win 11 on a Ryzen 7 PC with 32GB of RAM but I'm using a sd card and I'm using an Ender 3 v2. Haven't found any way of fixing it but I can't even use Cura anymore because it crashes.

ryhola, 2023-08-15:
Not sure if this will work for you but I updated all of my drivers and reinstalled Cura. Now it works perfectly again.

fieldOfView, 2023-10-02:
Closing this as a duplicate of #11608 (which @GrahamDyer says is the same issue).

closed 2023-10-02 by fieldOfView

GrahamDyer, 2023-10-31:
Cura 5.5 is basically unusable, 10GB RAM usage in a few minutes. 

https://github.com/Ultimaker/Cura/assets/3287001/13262321-e9bb-4eec-8fde-cdfc92c9ae8f