Ultimaker/Cura #16478: Unexpected error, 0.15mm nozzle

sanslos, 2023-08-14:
### Cura Version

5.4.0

### Operating System

Windows 10 Pro

### Printer

RadZ 0.15

### Name abnormal settings

Extra fine - 0.06mm, using 0.15mm nozzle

### Describe model location

Flat, centered

### Describe your model

Hollow christmas decoration, room for light.  Printed before with 0.4mm nozzle

### Add your .zip here ⬇️

[CFFFP_Christmas_Village.zip](https://github.com/Ultimaker/Cura/files/12339471/CFFFP_Christmas_Village.zip)

sanslos, 2023-08-15:
Note that I was attempting to slice the full-size model to get a feeling for how much to scale it for a reasonable print time, given the nozzle size.  At the moment I'm running a 50% scaled version which sliced without issue. The print time is estimated to be 9 hours 27 minutes.

The 0.15mm nozzle is capable of extremely precise prints (with a lot of tweaking), but the models must necessarily be small.  It often difficult to balance a meaningful print time with a good sized print.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic