Ultimaker/Cura #16487: [5.4.0] Unable to slice a specific model from fusion 360

jeremie5, 2023-08-15:
### Cura Version

5.4.0

### Operating System

Windows 10, 3080Ti, 13900k

### Printer

Tevo Black Widow modded to SKR1.4

### Reproduction steps

I have tried all file formats available in fusion 360.
Mesh Tools says mesh is watertight.

### Actual results

Failure to slice

[image]

### Expected results

Successful slicing

### Add your .zip and screenshots here ⬇️

[files.zip](https://github.com/Ultimaker/Cura/files/12346097/files.zip)

jeremie5, 2023-08-24:
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic