Ultimaker/Cura #16494: Some models cause the printer to stop mid-print

Unicod3r, 2023-08-16:
### Cura Version

5.4.0

### Operating System

Arch Linux

### Printer

Ender 3 V2 with mriscoc's firmware

### Reproduction steps

1. Start the print using one of the pinned .3mf files.
2. Wait until the printing stops.

### Actual results

The printer/Cura confirms the completion of the printing process, but the resulting object is done for only 50%.
[image]

It doesn't matter if the model is printed using USB or SD card.
I also checked the printer's settings: I found nothing that could cause this behaviour.

### Expected results

A complete model.

### Add your .zip and screenshots here ⬇️

[The .3mf files that fail to print](https://github.com/Ultimaker/Cura/files/12359943/debug.zip)

GregValiant, 2023-08-17:
Thanks for the report.
This isn't the first time this behavior has come up and it has never been a problem with the gcode, and so it has not been a problem with Cura.  When I slice your projects there is no problem with the gcode and I am confident that they would print correctly on my Ender 3 Pro.

- When you print via the USB using Cura's "USB Printing" plugin, there can be a problem with buffer over-run in the printer.  Dependent on the firmware - that is usually characterized by "Click to continue" showing up on the LCD screen of the printer.  It is possible to get around that by using the "Arc Welder" plugin in Cura, or by adjusting the "Mesh Tools/Maximum Resolution" to a higher value.  The default for most Ender is 0.4.
- When printing via the SD card - if the card has corrupt memory sectors the operating system doesn't know that and can write a portion of the gcode to a bad sector and the printer can't read it so the print stops for no apparent reason.  Re-formatting the SD card will usually fix that problem.  If the card is really bad it would need to be replaced.
- Since a common factor in SD printing and USB printing is the printer itself, the problem could be the firmware, or the printer mainboard fan isn't working, or some other hardware problem.

This is one of those things that if it were a Cura problem it would happen to everyone and nobody would use the software.

I'll leave the bug label on this for now.  Copy your gcode file from the SD back to the hard drive and then zip the file and post the zip folder here.  All I would do is open it in a text editor and see if it is complete and ends with "End of Gcode" or if the code quits abruptly in the middle of a layer.

Unicod3r, 2023-08-17:
I'm sorry for applying the wrong issue type.

Here is the .gcode file:
[CE3E3V2_catapult-PartPiston.gcode.zip](https://github.com/Ultimaker/Cura/files/12371025/CE3E3V2_catapult-PartPiston.gcode.zip)

It does end properly. Right now I'm checking for differences between original gcode and the gcode from the SD card.

Unicod3r, 2023-08-17:
The files are identical.

GregValiant, 2023-08-18:
That's hard to do because the E values are to 5 decimal places and there will be rounding errors.

I don't see any problems in the gcode you posted.  At ;LAYER:3 in the gcode there is an M106 line for the fan.  Between that and the end of the file there are no "M" commands at all.  There are comment lines (ex: ";TYPE:SKIN") and there are G0 and G1 lines, and that's it.  So there are simply no commands that could halt or pause a print.

Unicod3r, 2023-08-18:
Weird... Then I will try to reset the printer's settings.

Unicod3r, 2023-08-31:
Looks like the issue is gone after resetting printer's settings.
Thanks for help, everyone.

Unicod3r, 2023-10-22:
Nope. I left another model print overnight and it did the same thing.

I attached an archive with the .gcode, .stl and .3mf files.
The print stopped ~16.3mm Z. I checked the layers near that height, but nothing special there.

Unicod3r, 2023-10-28:
I was experimenting with the mriscoc's firmware and the issue is gone on all models now.
Maybe the update fixed it, or I had some very sneaky bit flip because of a faulty SD card.

I hope the issue does not come back.

closed 2023-10-28 by Unicod3r