Ultimaker/Cura #16495: Doesnt print continued line in a circle/round objects

gurkenfolie, 2023-08-16:
### Cura Version

5.4.0

### Operating System

Win10

### Printer

Anycubic Mega S

### Reproduction steps

When printing the line is not printed circular, e.g. you have a tube with 2 lines width it doesnt print circular without a stop, instead it stops after a 360 degree line and prints the other direction for 360 degrees (clockwise - counterclockwise).

It would save some time if the printing order would be circular in one direction (e.g. all 2 lines clockwise) instead of changing the direction as the direction change requires slowing down and is prone to produce unregularities. vase mode does this but only for vase mode models. 

### Actual results

direction will change after each 360° line 

### Expected results

all lines in one direction if printing circular

### Add your .zip and screenshots here ⬇️

[test tube.zip](https://github.com/Ultimaker/Cura/files/12360622/test.tube.zip)

GregValiant, 2023-08-17:
Thanks for the report.
As the nozzle moves along the inside diameter and is moving CW there is air on the right of the nozzle and model on the left.  When the outer wall is extruded then yes, the direction appears reversed and the ID is printed CCW BUT there is air on the right and model on the left.  So the nozzle remains oriented to the part the same on both the ID and the OD.  This is the current expected behavior and is the way it was designed.  I'll go ahead and remove the bug label.

I understand what you are saying but I think this should be a "feature request".

gurkenfolie, 2023-08-17:
Ok, thanks, I thought this is unintentional behaviour an so a bug.

Can you explain what you mean with air right of the nozzle? I dont really understand it. And I dont understand the advantage of this behaviour.

GregValiant, 2023-08-17:
It's like looking out the windshield of a car...if you were to sit behind the nozzle and ride along - the nozzle is in the right lane so the "shoulder" of the road is the air in this (poor) analogy and the shoulder is on the right.
When the nozzle moves to the OD it does a U-turn so that it is in the right lane with shoulder (air) on the right.
That can actually change if there were inner walls to deal with.  I find it to be consistent to itself, if not terribly easy to understand.

gurkenfolie, 2023-08-17:
Thanks, now I got it, Good explanation. Is there any advantage to this behaviour over "normal" lines?

I just see disadvantages: longer print time due to direction change and the "reversed" line will be extruded directly beneath the just printed line, so the just printed line has not much time to cool down.