Ultimaker/Cura #16497: great lion wall light

alexdesroches89, 2023-08-16:
### Cura Version

5.3.1

### Operating System

windows 11

### Printer

anycubic kobra

### Describe your model

it's the only model the slice on cura didnt work

### Add your .zip here ⬇️

[Great_Lion_Wall_Light_0d33f4a0ca.zip](https://github.com/Ultimaker/Cura/files/12363064/Great_Lion_Wall_Light_0d33f4a0ca.zip)

alexdesroches89, 2023-08-16:
cura version is 5.4 and not the 5.3

GregValiant, 2023-08-17:
Thanks for the report.
That model has been here before and it has errors in it.  Since this report is identical to #16366 I'll mark this as a duplicate and go ahead and close it.  I hope you understand.

mksbrasil, 2023-12-08:
at the end i got it sliced and printed in two parts after repairing the model manually and with the help of Meshinspector.