Ultimaker/Cura #16498: Slice failed CURA 5.4.0 but work with CURA 5.3.0

LECODEBARRE2224, 2023-08-16:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 S1 Plus

### Name abnormal settings

aucun

### Describe model location

jsp

### Describe your model

jsp

### Add your .zip here ⬇️

[fichier.zip](https://github.com/Ultimaker/Cura/files/12363359/fichier.zip)
[image]
cdt
Mr Delahaye

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

LECODEBARRE2224, 2024-06-11:
Hello Ultimaker Cura team!

I thank you for your feedback ! Yes indeed there were new versions of Cura
which were released and I tested my 3D object and there was no slicing
problem! :-D Thank you for your help and your feedback, I apologize for my
late response, I had not seen your email.

Good luck to you ! Sincerely,

Mr Delahaye William