Ultimaker/Cura #16499: Error Sicer

Fainen69, 2023-08-16:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Kingroon KPS Pro V2

### Name abnormal settings

no

### Describe model location

è già al centro

### Describe your model

nella versione 5.3.1 funziona correttamente

### Add your .zip here ⬇️

[KK_Portapenne_Arbeia v6.zip](https://github.com/Ultimaker/Cura/files/12364085/KK_Portapenne_Arbeia.v6.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

Fainen69, 2024-04-03:
...I downloaded and now everything works. Thanks for your help