Ultimaker/Cura #16512: Tree Support Base Ring Printing and Peeling

OriginalDrjinG, 2023-08-18:
### Cura Version

5.4.0

### Operating System

Win11

### Printer

Creality Ender3 Pro

### Reproduction steps

Create print with tree support, 
Max Branch angle: 51.0
Support Placement:  Touching buildplate
Support overhang angle:  51.0
Support Pattern:  Lines
Support Wall Line Count:  1
Enable Support Brim:  <checked>
Support Brim Line Count:  10

Build Plate Adhesion Type:  None
[CE3PRO_Belt_Hanger_2inch.zip](https://github.com/Ultimaker/Cura/files/12383494/CE3PRO_Belt_Hanger_2inch.zip)

### Actual results

Creating a print using Tree support causes a 1 line wide circle in the middle of the base of the support.  Using brim, this is fine because the space between the support exterior and this interior circle is filled in with material.  However, without a brim, there is no material to help with adhesion, the circle lifts up and attaches to the nozzle, which then drags and pulls up other base layer features.  I don't want to have to print with brim for every design...

### Expected results

Brim Support should fill in between inner and outer circle (support wall).

### Add your .zip and screenshots here ⬇️

[TreeTroubles.zip](https://github.com/Ultimaker/Cura/files/12383484/TreeTroubles.zip)

Asterchades, 2023-08-19:
This is a known issue, which should be resolved in the next release of Cura. Until then you need to use any option _other_ than "None" for Build Plate Adhesion if you want a Support Brim to print, or revert to an earlier version from before the fault (5.1 reportedly does not have the issue, I'm unsure about 5.2, and 5.3 onward have the problem). Unfortunately I cannot provide even an estimate as to when that release may occur.

Duplicate of #14983.

closed 2023-08-19 by Asterchades

OriginalDrjinG, 2023-08-19:
Thank you, and sorry for the duplicate.  I did do a search, but nothing stood out as the same issue.  I guess I didn't see 14983.  Cheers!