Ultimaker/Cura #16515: slicing fails when print thin walls turned on on this model

greencheetah100, 2023-08-19:
### Cura Version

5.4.0

### Operating System

win 10

### Printer

ender 3 pro

### Name abnormal settings

print thin walls

### Describe model location

slices on left side but not right or right-side up, moving did nothing

### Describe your model

semi truck badly converted from a mesh full of holes. first time trying to slice

### Add your .zip here ⬇️

problem was solved by 'lay flat' feature
[semi (1).zip](https://github.com/Ultimaker/Cura/files/12385971/semi.1.zip)

MariMakes, 2024-01-18:
hI @greencheetah100,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 8 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here https://github.com/Ultimaker/Cura/issues/16417

closed 2024-01-18 by MariMakes

HellAholic, 2024-02-16:
Small update,
Issue is resolved in CURA-11570 and will be part of the next Cura release.