Ultimaker/Cura #16518: Scaling issue

rventura22, 2023-08-19:
### Cura Version

5.4.0

### Operating System

windows 11 

### Printer

all

### Reproduction steps

Just updated software and when i try to load files into cura slicer, it scales all models to the max and to fix this, i always have to hit "reload models" under files tab.

### Actual results

Models load in at max scale

### Expected results

larger scale models than what they should be.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-08-20:
Thanks for the report.
I have had this happen but it has always been with models that I modified in MS 3D Builder and then brought back into Cura.  I think _that_ problem is within MS 3D Builder but it isn't consistent and so it's hard to nail down exactly what's going on.

Do you happen to have the "Mesh Tools" plugin loaded?  It has an automatic scaling utility that will (for example) auto-scale models designed in imperial units, to metric units.  It would be under the "Extensions" menu and then Mesh Tool Settings.  If the setting is off you could get consistent incorrect results.

If you happen to know one of the dimensions of the model (for example if you bring in a 20mm cube) then you could set the dimension in the "Scale" tool in Cura.  At that point if you note the percent scale it may point to where the problem is.

The Cura team will take a look.  If this was happening a lot there would be more reports and this is the first I've seen where it has been a repeatable problem.  Unfortunately, it's repeatable on your system but may not be on other systems.

JackTradz, 2024-01-19:
no im having same problem except it does that when i put it in barbarian but otherwise the prints come out a little smaller than suppose