Ultimaker/Cura #16520: Slicing failed

Buonojohn, 2023-08-19:
### Cura Version

5.4.0

### Operating System

MacOS Ventura 13.4.1

### Printer

Kobra Neo

### Name abnormal settings

none

### Describe model location

Enter of plate

### Describe your model

additions to gridfinity

### Add your .zip here ⬇️

[AK_hex_cups.3mf.zip](https://github.com/Ultimaker/Cura/files/12386491/AK_hex_cups.3mf.zip)

Buonojohn, 2023-08-19:
This bug only occurs if you multiply an object then slice.  I have consistently done it.  The project becomes unusable. But if you import the object twice the error does not occur.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic