Ultimaker/Cura #16529: [5.4.0] Print via USB doesn't home when it isn't the first print

DDdotDE, 2023-08-21:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3 S1 pro

### Reproduction steps

0. have print finshid (via sd)
1. Move Printer bed forward (to get print)
2. plug in USB-cable to printer and PC
3. send gcode via "print via USB"
4. proceed to work on your pc without checking what the printer does (it should just do its job, right?)
5. jump up and look after your printer after hearing a frightening "brrrrrrr"(the sound which the printer makes if sth. moves to far/ out of its limits)
6. watch your printer driving its head in the end of the bed (where the cables to the bed for an E3 s1 Pro are) while printing the pre extrusion

### Actual results

The Printer doesn't home before printing, thus driving its head in the end of the bed (where the cables to the bed for an E3 s1 Pro are) while printing the pre extrusion
[image]

### Expected results

The printer homes and prints the pre extrusion on the bed, not on cables

### Add your .zip and screenshots here ⬇️

[Linia L 2 kurz.zip](https://github.com/Ultimaker/Cura/files/12395382/Linia.L.2.kurz.zip)

GregValiant, 2023-08-21:
Thanks for the report.
Some changes were made to USB printing and rather than upgrade a version that worked for a lot of people, it appears that a decision was made at Ultimaker to include a second version.
By default, the plugin "USB Printing 1.0.2" is installed.  The alternate version "USB Printing NG 1.0.4" is available in the Marketplace.
Go to the Marketplace and click on the Gear icon and scroll down and disable "USB Printing 1.0.2" and then click on the "Plugins" button and scroll down and install "USB Printing 1.0.4".  At this point you must close Cura and restart.

I can't duplicate your problem with either version.  The first print from the SD finishes and then I print over the USB and it works fine.

One problem with trying to print over the USB is that there are so many hardware versions and so many software and firmware versions and 4 or 5 different operating systems involved that getting USB printing to work with all of them is tough if you don't have access to every one of the combinations.  So what you get is the generic approach as "One size fits most".

DDdotDE, 2023-08-22:
Thank you for your reply, I'll look into that and try it out.
Side note, i looked a bit into the gcode and did some researchabout G28(auto home).
the gcode is `G28` O ; Home all untrusted axes` , which made me think, that the problem might also be with creatilliys firmware which might flag the y axis as trusted (or at its home position in my understanding) thus not trying to home it, which is wierd becauswhile printing via sd, it doesn't have this problem.
Maybe it has sth to with starting the print over usb, while tha printer is still at its "Finished screen"

GregValiant, 2023-08-22:
I had forgotten about that one.  Not all firmware supports G28 with the "0" parameter.  I put in a pull request to have the "0" removed.  The change should show up in the next version of Cura.

In Cura go to Manage Printers / Machine Settings and in the StartUp Gcode box change that line to a simple "G28" with no parameters.

DDdotDE, 2023-08-23:
yep, that fixes the problem and doing a full homing doesn't even take much longer.

GregValiant, 2023-08-23:
Alright!
Since it was a definition/startup gcode problem I'll go ahead and close this.  The pull request should fix the problem in the next release (if it makes it in under the deadline).

closed 2023-08-23 by GregValiant