Ultimaker/Cura #16535: brim printing on inside edge when outside only check box was ticked. using ultimaker 2+

Rand2mor, 2023-08-22:
### Cura Version

5.4.0

### Operating System

windowes 10

### Printer

ultimaker 2+ no mods

### Reproduction steps

did not want brim on inside edges so ticked outside only check box.

### Actual results

printed brim on inside edge around hole in the middle but not the smaller holes near the outside edges

### Expected results

not print brim, on inside edges

### Add your .zip and screenshots here ⬇️

### none taken

StmLord, 2023-08-22:
I can confirm, brim only on outside does not work on 5.4.0 Mac version.
I include the project file for further investigation

[brimProb.3mf.zip](https://github.com/Ultimaker/Cura/files/12413215/brimProb.3mf.zip)

Asterchades, 2023-08-23:
Only the first layer is taken into consideration when it comes to generating the brim. Due to the gap between the "teeth" on those Commodore logos this means there actually is no "inside" to speak of on the first few layers - everything is technically outside, even if the above layers go on to enclose it.

If you flip them on the Y axis, enclosing the first layer, there is no brim generated internally.

[image]

MariMakes, 2023-08-24:
Hey @StmLord,

Thanks for the project file, I agree with @Asterchades that there is no inside and outside in the original orientation because it's not an enclosed O shape but more an opened C shape. 
[image]

@Rand2mor,

Welcome to the UltiMaker Cura Github 🚀 
I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Rand2mor, 2023-08-31:
> On ,Wed Aug 23 2023 15:20:54 GMT+0930 (Australian Central Standard Time),

the job i did was a large rectangular section with a  12mm hole near the middle of it.  it printed a brim inside the hole as well as on the outside

MariMakes, 2023-09-04:
Hey @Rand2mor,

To help you, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-09-18 by github-actions[bot]