Ultimaker/Cura #16542: skirt prints wrong

Rand2mor, 2023-08-23:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

ultimaker 2+ no mods

### Reproduction steps

set skirt line count to 0 by mistake.  i meant to set skirt line count to 1
set skirt distance to 3
then sliced and printed

### Actual results

skirt printed at no distance from job  and printed 1 line for the first 3 layers

(i have noticed before that the skirt when it prints, it prints 1 skirt line for the first (skirt distance setting) number of layers)

0

### Expected results

dont know as setting skirt line count to 0 was a mistake.  By setting skirt line count to 0 and getting  the result i got shows that you have mixed up the skirt line count setting with the skirt distance setting.   And the skirt line count is printing the lines in the vertically direction instead of in the horizontal direction.

### Add your .zip and screenshots here ⬇️

nothing to add

Asterchades, 2023-08-24:
This is expected behaviour when considering the default settings - though the settings in question may not necessarily be visible in your UI, depending on your configuration.

The first is `Skirt/Brim Minimum Length`. This defaults to 250mm and will override the `Skirt Line Count` setting in order to ensure that minimum length is reached. This is why setting the Skirt Line Count to 0 still resulted in a Skirt being printed. This setting is not visible until Expert unless manually toggled.

The second is the `Skirt Height` setting, which is new to 5.4. This defaults to 3 layers and pretty much does exactly what it says it will, making the Skirt taller. The intent is to make it easier to remove without resorting to using a tool or fingernail, potentially causing damage to the print bed or yourself. These vertical layers are not taken into consideration for fulfilling the `Skirt/Brim Minimum Length`. Again this setting is not visible until Expert unless manually toggled.

Rand2mor, 2023-08-31:
> On ,Thu Aug 24 2023 14:42:38 GMT+0930 (Australian Central Standard Time),

thank you for your explanation.  i had not read up on the new changes but it is a good idea for the reason given. thanks for the reply

cheers  Robin