Ultimaker/Cura #16543: Slicing of Voronoi pot STL failing on Cura5.4

monticellifernando, 2023-08-23:
### Cura Version

5.4.0

### Operating System

Linux

### Printer

Ultimaker S7

### Name abnormal settings

It crashes with default settings

### Describe model location

It does not slice rotating nor moving it around

### Describe your model

I did slice this model with previous versions of Cura and printed it. I didn'r try Mesh Fix with Meshtools Plugin

### Add your .zip here ⬇️

[UMS7_MacetaVoronoi_3.3mf.zip](https://github.com/Ultimaker/Cura/files/12420359/UMS7_MacetaVoronoi_3.3mf.zip)

The crash happens with this stl I made some time ago: https://www.thingiverse.com/thing:5267104 
I managed to slice and print the object with my custom printer with older version of Cura back then.

I was notified by a User few days ago they couldn´t slice it. I just tried and confirmed cant be sliced with Cura 5.4

5axes, 2023-08-23:
Standard Issue , Initial bottom surface not really flat.

Move the model fro Z -0.001 to Z-0.01

[image]

monticellifernando, 2023-08-23:
Oh! Thanks! I guess I could just modify the model and make the bottom flat

5axes, 2023-08-23:
Unfortunately, it's not so simple... By modifying the **_Horizontal Expansion_** and **_Initial Layer Horizontal Expansion_** parameters, the problem may be still present.  This workaround is just a temporary solution not THE solution.

monticellifernando, 2023-08-23:
Oh ok. I'll try anyways with a flatter base modification for this and see if it works. At least until this bug is fixed.

happyme531, 2023-08-24:
[CFFFP_Core.zip](https://github.com/Ultimaker/Cura/files/12431233/CFFFP_Core.zip)
Same issue, Horizontal Expansion -0.3 -> Fail, 0.0 -> Okay

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic