Ultimaker/Cura #16549: [5.4.0] Machine End G-Code Not Being Written in File

mzhu1113, 2023-08-24:
### Cura Version

5.4.0 - Though this has happened in earlier version

### Operating System

Windows 11

### Printer

NA

### Reproduction steps

Please note: I'm unable to provide the 3MF file for this because this job is under NDA.

Unfortunately I do not have the reproduction steps because a reslice of the 3MF project file (we save this after the gocde file save) will have the Machine End G-Code despite it being saved right after the offending gcode was already produced. You can see this in the screenshot of the generated gcode. This leaves the printer hovering over a part and with all heaters and fans on. Luckily this is not a terribly large of an issue for us because we have employees that will get to the printer before damage can be done to the part (or printer) but maybe not for a hobbyist.

**The issue is not apparent until after the job has been run.**

Additional Notes:
This rarely happens but it does occur once out of every 3-4 months and we process about 200-300 unique jobs a month. I apologize for not having more than just a smoking gun of the issue. One thing I do know is that we can have an instance up and running for a while and do swap between printers for certain jobs to maximize efficiency.

### Actual results

No Machine End GCode is written to the gcode file.

### Expected results

Machine end gcode to be written to the gcode file.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-08-24:
Thanks for the report.
Cura will typically add a couple of other lines before the Ending Gcode goes in.  

;TIME_ELAPSED:15276.127562 >>> Total Print time estimate.
;MESH:NONMESH >>>> >>>>>>>This may or may not be added depending on your Combing setting.
M140 S0 ; >>>>>>>>>>>>>>>>>Turn off Hot End
M107  ;>>>>>>>>>>>>>>>>>>>Turn off fan.
;                           >>>>>>>>>>>>>The start of the actual Ending Gcode.
;Greg's Ending
;

The integer portion of the last "TIME_ELAPSED line in the file should match the "TIME" line that is in the front of the file.  In the case of this example it's:
;FLAVOR:Marlin
;TIME: **_15276_**
and
;TIME_ELAPSED: **_15276_** .127562
So you could tell if just the Ending Gcode was missing or if some other portion of a Layer was missing.  That's something the Cura team will want to know.
Do you happen to save directly from Cura to an SD card?  Are you printing over a network, using the SD, or USB Printing, or Octoprint?
There are known cases where a portion of a gcode file was written to a bad sector on a memory card.  It looks much like your example looks - cut off in the middle of something.  I'm not saying that's what happened, but it's something that can happen.

FYI: a real oddity is when the back portion of one gcode gets written to a bad sector and the front portion of another gcode gets written to the same bad sector.  The printer doesn't see the EOF of the first file and happily moves right into the second file which is missing it's Start Of File marker.  You get the bottom of one model and the top of different model. 

In regards to the NDA - You are a power user and I'm sure the Cura team ( maybe @MariMakes ?) could make arrangements to move this to E-mail messages instead of here on Github.  Would that be acceptable?

mzhu1113, 2023-08-25:
Hey @GregValiant 

Thanks for responding.

> So you could tell if just the Ending Gcode was missing or if some other portion of a Layer was missing. That's something the Cura team will want to know.

Ah yes, my apologies on forgetting to mention this part. Both New and Old (bad) final gcode files show it is printing the last object on the plate at the last layer and last coordinates. However on the bad gcode it ignores the retraction, M140/M205/M107 commands, and the Machine End G-Code.

Attaching the new and old (left and right respectively) screenshot.
[image]

[image]

Please note: This is the specific behavior that happens and it's very very rare. The bad sectors issues we typically catch by cycling out the SD Card (marking the suspected bad one) and seeing if the 2nd attempt works, if so, we e-waste the previous SD Card. If the 2nd attempt fails, then we check the Gcode file itself.

>Do you happen to save directly from Cura to an SD card? Are you printing over a network, using the SD, or USB Printing, or Octoprint?
Local Save > Auto Saved to Network Drive > then it is loaded onto a printer (SD Card) by a Printer Tech (current process).
The file that was saved locally had this issue.

We only load one file at a time for process reasons so we don't experience the interesting situation you're mentioning. Would be funny to come in the morning to see that haha.

>In regards to the NDA - You are a power user and I'm sure the Cura team ( maybe @MariMakes ?) could make arrangements to move this to private E-mail messages instead of here on public Github. Would that be acceptable?
This will work though I'm not sure how helpful it'll be. 

Thanks!

GregValiant, 2023-08-26:
Thanks for that information.
Someone from the Cura team will take a look.  Right now the information is sparse so I don't know if they will have any luck chasing this down.
A post processing script could look for the line ";End of Gcode" and pop up a message if it's missing.  If it is missing it could also go ahead and write in the Ending Gcode.  You would need to have it running all the time and it would need to be last in the list of post-processors.  For now I think that's all you can do.

I'm just the greeter here but I dabble in post processors.  I'm not sure how this problem could occur.  My understanding is that the Ending Gcode is inserted by the Cura Engine prior to any post-processing, and that the "SETTING_3" section is inserted after any post-processing.  How both of the last two items of the data list come up missing is a mystery to me.
The other fact is that the data is conveniently cut off right at the end of the last layer and that would seem to indicate that it is a software problem, or, it is an amazing coincidence that is hiding a hardware problem.

Maybe @nallath can straighten me out and shed more light on this.

MariMakes, 2023-08-28:
Hey @mzhu1113,

Thanks for your report 👍
I would agree with @GregValiant that it's hard to troubleshoot with a project file.

If you submit a case via our official Support Channel and mention my name, we can securely exchange some files. 
You can submit a case here: [https://support.ultimaker.com/s/contactsupport](https://support.ultimaker.com/s/contactsupport)

I can ask the team for details on where to start troubleshooting. 
But I expect that if we don't have proper steps to reproduce, we can't help with resolving the issue. 
Fingers crossed 🤞

mzhu1113, 2023-08-28:
> Hey @mzhu1113,
>
> Thanks for your report 👍 I would agree with @GregValiant that it's hard to troubleshoot with a project file.
>
> If you submit a case via our official Support Channel and mention my [...]

Will do and thanks! I'll reach out in the next couple of days. Got to get through a few priority jobs in the queue.

MariMakes, 2023-09-01:
Hi @mzhu1113,

I showed your issue to the team. 
They mentioned that the next release will have a warning if a file was not completely written, probably because you ran out of memory or space for your gcode of ufp.  https://github.com/Ultimaker/Cura/pull/16556

Because we've received reports, like https://github.com/Ultimaker/Cura/issues/14273, that if there is not enough space that the file would just be incomplete.

We think it might be related because your issue doesn't happen consistently 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

mzhu1113, 2023-09-04:
> ecause we've received reports, like #14273, that if there is not enough space that the file would just be incomplete.
>
> We think it might be related because your issue doesn't happen consistently
>
> To [...]

For me personally and for the team here some kind of warning is enough so I believe this case can be closed. I don't think it's completely reasonable to chase such a rare edge case like this when it happens once after thousands of gcode saves. Especially since it might even be machine specific 😅 

Thanks again.

closed 2023-09-04 by mzhu1113