Ultimaker/Cura #16552: Skirt height setting doesn't work with support enabled

TomN7, 2023-08-24:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Flashforge Adventurer 4

### Reproduction steps

1. Imported a model into Cura.
2. Enable skirt with height set to 3 layers
3. Enable supports
4. Slice Model

### Actual results

The skirt does not have additional layers printed, and remains one layer high.
The first section of support printed has an extra outline printed instead.

### Expected results

The skirt should have additional layers printed, and the supports should print as normal.

### Add your .zip and screenshots here ⬇️

[image]

MariMakes, 2023-08-28:
Hi @TomN7,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

[We've already seen the report on our community](https://community.ultimaker.com/topic/43823-ultimaker-cura-54-stable-released/?do=findComment&comment=325036) so I discussed it with the team last week. They agreed that it is a bug, especially since the layers seem to be printed twice. 

[image]

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10968

Thanks for the report! 👍

MariMakes, 2023-10-11:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-11 by MariMakes