Ultimaker/Cura #16559: Consider the layer following the infill as a bridge

HanSolo30, 2023-08-25:
### Is your feature request related to a problem?

When I use PETG, the first layer after an infill pattern like grid or cubic, fails.

### Describe the solution you'd like

Other softwares, like PrusaSlicer, consider that layer as a bridge, and allows to select a different speed. With low speeds, the layer works fine.
Using PETG at the same speed as the rest of the print, rises the material.

### Describe alternatives you've considered

An alternative can be a speed setting for the first layer after the infill.

### Affected users and/or printers

Users using petg and similar materials.

Asterchades, 2023-08-26:
I believe this can already be done, though the method isn't necessarily as straight forward as a single button.

When you enable the `Enable Bridge Settings` option in Experimental, one of the settings it uses is called `Bridge Sparse Infill Max Density`. This setting determines whether the Bridge settings are applied over Infill or not - any area which has _less_ Infill than the percentage here (default 0%) will be considered a Bridge.

HanSolo30, 2023-08-26:
I tried with no results. Maybe with incorrect parameters. I will try again.