Ultimaker/Cura #16560: Expansion horizontale

Aquila13fr, 2023-08-25:
### Cura Version

5.4.0

### Operating System

Windows 10 Famille

### Printer

Many

### Reproduction steps

If i put an horizontal expansion, the slicer cut a horizonal surface creating holes.
[image]
[image]

### Actual results

Print with holes...

### Expected results

I need print without holes

### Add your .zip and screenshots here ⬇️

[Haut.zip](https://github.com/Ultimaker/Cura/files/12439010/Haut.zip)

Asterchades, 2023-08-26:
There appears to be two settings working in tandem here resulting in the fault - `Horizontal Expansion` and `Skin Expand Distance` (I would have guessed this was a Removal fault). Adjusting just one of them in isolation doesn't cause a problem, but when both are adjusted together things start to go wrong. Adjusting `Skin Removal Width` on top then goes on to make things worse.

By a staggering co-incidence it seems to be that 1.6mm of `Skin Expand Distance` is exactly the right (wrong?) value for this particular model. 1.5 or below works fine, as does 1.7 or above, though the actual limits are closer to 1.6 and vary depending upon what the `Skin Removal Width` is set to (higher values for Removal seem to shift the limits higher as well).

Note 1: This project doesn't specifically adjust `Skin Expand Distance`. However it does increase the `Wall Count` to 4, which in turn increases the `Wall Thickness` to 1.6mm, which subsequently increases both Skin Expand and Removal accordingly.

Note 2: While 3D Builder reports this model as having faults, correcting them does not alter the (mis)behaviour.

@MariMakes Got a weird one here for you!

MariMakes, 2023-08-28:
Hey @Aquila13fr ,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report and thanks for tagging me @Asterchades 👍

It reminds me of https://github.com/Ultimaker/Cura/issues/12425 and https://github.com/Ultimaker/Cura/issues/15028 🤔 
Those issues could also be resolved by adjusting the skin expand distance.
I hope that helps you get unstuck for now✌️

We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12425

Thanks for your report 👍

closed 2023-08-28 by MariMakes