Ultimaker/Cura #16563: Killed SD card after ejecting

trainmanztr, 2023-08-26:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Ender 3

### Reproduction steps

1. Slice File
2. Save to removable drive
3. Eject using the built in eject button

### Actual results

Micro SD card is completely killed
Can't even reformat it

### Expected results

Should be able to use Micro SD card like normal

### Add your .zip and screenshots here ⬇️

[image]

trainmanztr, 2023-08-26:
IT JUST HAPPENED AGAIN BUT I DIDNT EVEN USE THE BULT IN EJECT BUTTON. THIS IS PISSING ME OFF. IT'S HAPPENED A COUPLE TIMES BEFORE THIS TOO!!!!

GregValiant, 2023-08-26:
Thanks for the report.
All the disk operations are handled by the operating system.  When you click the button:  Cura passes the request (to save or eject) to the Windows API.  That's how virtually all software works.  So Cura itself never actually interacts with the disk.
Maybe there is an issue with your card reader, or something might be wonky with Windows.
_"...BUT I DIDNT EVEN USE THE BULT IN EJECT BUTTON..."_  Could be a clue?
I'll leave the bug label on this for the Cura team to look at as there have been other issues with Windows11...but for your own troubleshooting - this is unlikely to be a Cura problem.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant