Ultimaker/Cura #16585: I want to request a function to "pause print at painted object"

DuckersMcQuack, 2023-08-27:
### Is your feature request related to a problem?

No problem at all, more a nifty feature for multi color prints

### Describe the solution you'd like

The solution would involve a new function to see model as faces of a model then have a "paintbrush" in different colors, to target certain features of a model to initiate pause command mid layer to change filament at the trajectory when the slicer indicates "that part's turn".

The color codes would be for flow, so no filament change is under nor overextruded during the print as it's then preset with that spool's flow to be the same accuracy all over the print.

the inside of the model would have a half sphere taking as many extra walls as print's parameter does in order to properly "wipe" the nozzle from excess oozing/old filament/new filament before starting on the outer wall with the actual color.

It would end up pausing a lot, but with user's effort, the print could look fantastic with many different colors.

### Describe alternatives you've considered

Manually editing Gcode to find out at what movement one would manually add pause command to change filament, which could be tedious.

### Affected users and/or printers

All non multi color printers

GregValiant, 2023-08-28:
If you have a single-extruder printer and want a multi-extruder type of print you could slice a set of models with a multi-extruder printer active and Cura would put Tool Changes in the gcode.  Then you have a post-processor convert each "Tool Change" to a "Pause with Filament Change".  Effectively it makes you the tool changer.

I've written such a post-processor.  Remember that there will be at least one filament change on every layer where there are multiple models.  It quickly loses it's appeal after the 87th filament change and there are 371 more to go.
It is the definition of "Tedium ad nauseam".

DuckersMcQuack, 2023-08-29:
Aye. But it would would not do multiple pieces of a model separately added, but simply "paint faces of a single model" 

Tedium indeed, but i'd go through with it if it can make my model cool as fuck lol

GregValiant, 2023-08-30:
I only printed one model with my plugin.  I did the regular method of having two models fit together in Cura, printer configured for dual extruders with shared heater and nozzle.  It worked well.  Each tool change gets converted to a pause with a filament change.
At the end of the print I was thinking "why did I do this to myself?".  There were 30 consecutive layers that had a filament change on each and the layers each took about 1/2minute to print so there were 30 changes in 15 minutes.

The trailer body on the left was with the "Emulate a Multi-Extruder" plugin, and on the right was with regular "Pause at Height.
[image]