Ultimaker/Cura #16587: makerbot portion of cura does not work

eboldt10, 2023-08-27:
### Cura Version

4.12.0

### Operating System

windows 11

### Printer

makerbot

### Name abnormal settings

no

### Describe model location

no

### Describe your model

base for a gnome model

### Add your .zip here ⬇️

[Rude_Cheesehead_Gnome_Poker_Card_Capper_-_Green_Bay_Packers_2523717.zip](https://github.com/Ultimaker/Cura/files/12449008/Rude_Cheesehead_Gnome_Poker_Card_Capper_-_Green_Bay_Packers_2523717.zip)

eboldt10, 2023-08-27:
when I try to slice for the makerbot, i get an error that says there are no formats to write with.

GregValiant, 2023-08-29:
Thanks for the report and the file.
We will need a project file to investigate this.  Load the model in Cura and set Cura up to slice then use the "File | Save Project" command to create  a 3mf project file.  Zip the file and post the zip folder here.
You might want to try a Makerbot  specific help group to see if anyone  has run into this problem previously.  Another thing would be to update Cura to a more recent release.  4.12 was 10 releases ago.  4.13.1 is very good if you don't want to install an Arachne version.

closed 2023-09-13 by github-actions[bot]