Ultimaker/Cura #16590: Slicing failed, Error code 3221225477

uoutac, 2023-08-28:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic Kobra

### Describe your model

I'm not too sure what I did to cause this, but nothing over 10 mb will slice, they all fail with a generic "Backend exited abnormally with return code 3221225477!". I am trying to print a gun frame, so I'm not sure If I'm allowed to share it on here. I will however provide a profile. (it will be a text file, so remove the extention before importing)

### Add your .zip here ⬇️

[uoutac_profile.curaprofile.txt](https://github.com/Ultimaker/Cura/files/12452188/uoutac_profile.curaprofile.txt)

GregValiant, 2023-08-28:
Thanks for the report.
My understanding is that it is a memory error.  Certain computers seem to be more prone to that failure mode and Windows 11 seems to be having some occasional issues with Cura.  Maybe @rburema has more information regarding the error.

rburema, 2023-08-29:
All I can say that this error is the generic segmentation fault one (`0xC0000005` in hex), so what we call a 'hard crash' in the engine. As such, it doesn't give much useful information unfortunately.

While I'm aware of problems on Windows 11, those seem to be related to the frontend, not the engine, as far as I'm aware at least.

It would be helpful to have to logs, just to be sure, but I don't expect much.

If nothing above 10Mb or so slices for you, but you're hesitant to share the original model, surely there is another model that you _can_ send? I can't promise anything, but we can at least check if the bug is reproducible on our side.