Ultimaker/Cura #16599: New tree suports huge bed sticking issue

ramsestom, 2023-08-28:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Artillery sidewinder X2

### Reproduction steps

I recently update from cura 5.2.x to cura 5.4.0 and I have a big problem with the way new tree supports are made.
The tree supports from the previous cura version where designed by cura with a plane and plain contact interface with the printer bed. So there was no risk of them detaching from the printer bed during the print. But in cura 5.4.0 the tree suports are directly printed on the bed with no base. It's just the wall of the branch (a single line) that is in contact with the printer bed. As a consequence when printing the second or third layer, these supports tend to come off the bed (even with a heating bed) and get caught by the printer nozzle, completely ruinning the print. Actually, I have not succeed to have a correct print with the new tree supports without using a raft (wich is a nonsense as I have a heating bed so it's just wasting material because of the baddly designed/non existing base of new tree supports).
Would it be possible to restore the plain contact base of tree supports with the bed, or at least have an option in the interface to force them without having to get a raft for the full print?    

### Actual results

tree supports first layer detaching from the printer bed during the print of next layers, completely running the print

### Expected results

tree supports sticking to the printer bed (by intreassing the contact interface)

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-08-29:
Thanks for the report.
Yes, there is a bug that affects the "Support Brim" and the "Prime Tower Brim".
The workaround is to enable "Skirt" or "Brim".  If you enable the skirt you can set the "Skirt Line Count" to "0" and there won't be a skirt but it will fool the bug that affects the Support and Prime Tower Brims.

I'll mark this as a duplicate and go ahead and close this.  The Cura team is aware of this bug and it has been worked on and should be fixed in the next release.  I hope you understand.

closed 2023-08-29 by GregValiant

ramsestom, 2023-08-29:
Thanks for the quick reply. 
The trick of the Skirt with a 0 line count (tested it today on a print and it works fine) would do the job for now until this bug is properly solved in the next release