Ultimaker/Cura #16602: chassis_scaled

REDTRUK, 2023-08-28:
### Cura Version

5.3.0

### Operating System

Windows 11

### Printer

TENLOG

### Describe your model

No - No - No

### Add your .zip here ⬇️

[LAV_25_1960505.zip](https://github.com/Ultimaker/Cura/files/12458781/LAV_25_1960505.zip)

GregValiant, 2023-08-29:
Thanks for the report.
That model won't slice correctly without repair.  A lot of areas are simply missing because of errors in the model.
This is an analysis of the model:
--> 218 Naked edges (?)
--> 6 Planar holes (?)
--> 1 Non-planar holes (?)
--> 3700 Non-manifold edges (?)
--> 2278 Inverted faces (?)
--> 238 Degenerate faces (?)
--> 4446 Duplicate faces (?)
--> 0 Disjoint shells (?)
[image]

Here is the model in Mesh Mixer.  The balls and arrows indicate errors.
[image]

I'll remove the bug label and close this.  There isn't any way to tell if a bug is affecting the slice, or if it's because the model is so bad.  I hope you understand.

closed 2023-08-29 by GregValiant

REDTRUK, 2023-08-29:
Thanks for the help

Closed #16602 as completed.