Ultimaker/Cura #16604: My slicer had an unexpected error and can't slice my stl file.

Talomh, 2023-08-29:
### Cura Version

5.2.2

### Operating System

Windows 10

### Printer

Creality Ender 2 Pro

### Reproduction steps

1. I opened up the STL file and attempted to slice I, but it just said, "unexpected error"
2. I tried to rotate and scale the model, but it said the same message

[image]

### Actual results

It said "Slicing failed with an unexpected error" 

[image]

### Expected results

It should fully be able to slice without any errors ocurring.

### Add your .zip and screenshots here ⬇️

[UMS5_Ghost mask teeth (5).zip](https://github.com/Ultimaker/Cura/files/12460039/UMS5_Ghost.mask.teeth.5.zip)
[image]

Talomh, 2023-08-29:
HELP PLS

GregValiant, 2023-08-29:
Thanks for the report and the file.
I didn't have any trouble slicing it.
I checked the model on formware.co/onlinestlrepair and received this report.
-> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **2 Non-manifold edges (?)**
--> **3 Inverted faces (?)**
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 16480ms ------
-> Vertex count changed from 52047 to 57964 **(+5917)**
-> Triangle count changed from 106126 to 120866 **(+14740)**

You can see there are only a couple of errors, but look at how many vertices and triangles had to be added.  I would suggest you repair the model.

There is a slicing bug in 5.4.0 that has been reported numerous times.  There are three main ways to fool it.

- Rotate the model in Cura.
- Move the model away from the midpoint of the Cura build plate.
- Change the Initial Layer Height by a little bit (like from 0.20 to 0.23).

One or a combination of those may get you past the bug.

I'll mark this as a duplicate and I'll go ahead and close this as the Cura team has a lot of examples of this behavior.  You can still post here and let us know if one of those workarounds helped.

closed 2023-08-29 by GregValiant

Talomh, 2023-08-30:
Thank you! I opened the file again and for some reason it's working?? But if it didn't work, how would I repair the model? How can I add vertices and triangles?