Ultimaker/Cura #16605: Machine Vise slicing failed

jcwizard86, 2023-08-29:
### Cura Version

5.4.0

### Operating System

macOS Monterey

### Printer

Anycubic Vyper

### Describe your model

Machine Vise, sliced successfully before on another machine 

### Add your .zip here ⬇️

[AV_Knurled_Handle_Vise.3mf.zip](https://github.com/Ultimaker/Cura/files/12461486/AV_Knurled_Handle_Vise.3mf.zip)

GregValiant, 2023-08-29:
Thanks for the report.
I was going to take a look but I received a message that the zip file was corrupt so I can't open it.

DVSVIDEO, 2023-09-03:
> Thanks for the report. I was going to take a look but I received a message that the zip file was corrupt so I can't open it.

There is nothing wrong with the zip file. The 3mf file is corrupt or does not contain a model.