Ultimaker/Cura #16609: Andromda failing slicing

LostCause001, 2023-08-29:
### Cura Version

5.4.0

### Operating System

Windows

### Printer

Kobra max

### Name abnormal settings

No changes to the model or settings have been made.

### Describe model location

model starts the slicing, then fails in any position

### Describe your model

I have printed this model before, just wanted to reprint it.

### Add your .zip here ⬇️

[Andromeda__easy_print__front_rear_2982466.zip](https://github.com/Ultimaker/Cura/files/12467832/Andromeda__easy_print__front_rear_2982466.zip)

This is the model that I have printed before and just wanted to print it again. But refuses to slice,

GregValiant, 2023-08-31:
Thanks for the report.
It really isn't possible to investigate this without a project file.  Use the File | Save Project command and zip the 3mf file and post the zip folder here.

LostCause001, 2023-08-31:
[Andromeda__easy_print__front_rear_2982466.zip](https://github.com/Ultimaker/Cura/files/12481430/Andromeda__easy_print__front_rear_2982466.zip)