Ultimaker/Cura #16616: Slicing fails on when trying to create tree supports and brim.

boehmap, 2023-08-29:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Ender 3 Pro

### Name abnormal settings

Seems to fail when selecting both tree supports and a brim. I've tried with smart and "not smart" brims. Works on brim only, or tree only, but not with both.

### Describe model location

Moving and rotating does not cause it to build. Slicing one model at a time does fix it. 

### Describe your model

This is an air duct for a new print head. It is watertight, and I have tried mesh fix. This model appears to slice fine sometimes, and fail others, with the exact same settings.

### Add your .zip here ⬇️

[CE3PRO_Duct 4010 p2_L.zip](https://github.com/Ultimaker/Cura/files/12470127/CE3PRO_Duct.4010.p2_L.zip)

This isn't blocking, but does seem to show an issue with trees/brims.
This failure is not deterministic. Running the same settings seems to fail about 75% of the time

LilBub, 2023-09-04:
I can confirm this.
Brim around the part but not around the Tree Supports, despite Support Brim enabled

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic