Ultimaker/Cura #16619: Slicing failed when trying to slice Star Destroyer from nikoli2803

FinDoo20, 2023-08-30:
### Cura Version

5.4.0

### Operating System

Win 11

### Printer

Tevo Tornado Gold Version

### Reproduction steps

Trying to slice Imperial Star Destroyer - Star Wars High Detail from nikoli2803. Ran the stl files through stl repair https://www.formware.co/onlinestlrepair, but did not fix the slicing issue. Although it did say that there were some issues with the stl files. 

### Actual results

It begins slicing than ends, displaying issue. 

### Expected results

Sliced; from stl to gcode. 

### Add your .zip and screenshots here ⬇️

[Repaired.zip](https://github.com/Ultimaker/Cura/files/12473780/Repaired.zip)

GregValiant, 2023-08-31:
Thanks for the report and the files.
I didn't have any trouble slicing this even though the Front and Middle still show as not being watertight.

5.4 has a slicing bug that can sometimes be fooled by:

- Move the models around in Cura.
- Rotate the models in Cura.
- Change the Initial Layer Height by just a little bit.

See if any of those work for you.  I'll mark this as a duplicate and go ahead and close this.  The Cura team is aware of this bug and is working to fix it for the next release.
[image]

closed 2023-08-31 by GregValiant

FinDoo20, 2023-09-01:
Thanks for replying.
Yes, another thing i forgot to mention :D was that when the files were fixed, they did slice. Only when I increased the scale from 100 to 200 or so did it fail. When I sliced the parts individually, the middle and the front stl files would generally display error. This was when trying to slice at an increased scale.

Thanks for the help!

FinDoo20, 2023-09-01:
Oh, and reading other comments, I found some peoples' issues were fixed when changing the version slicer. I'll see how it goes later on

GregValiant, 2023-09-01:
Of the 5.x versions 5.2.2 is probably the best.  I use 4.13.1 for a lot of stuff.
When you scale a model up you also scale up any errors.  For example if a seam is open but not enough for Cura to notice it won't be a problem.  When you double the size of the model the gap in the seam doubles as well.  That might push it above what Cura can handle.
The last couple of releases of Cura have been troubled by a slicing bug.  It's being worked on and hopefully it will be fixed in the next release.

FinDoo20, 2023-09-01:
Thanks for the response, ill look into ver 5.2.2! Thanks