Ultimaker/Cura #16621: [5.4.0] Questionable end gcode in some default profiles for Anycubic ptiners

foosel, 2023-08-30:
### Cura Version

5.4.0

### Operating System

OS independent, this is a metadata issue

### Printer

Anycubic Vyper, Anycubic i3 Mega, Anycubic 4max

### Reproduction steps

Take a look at the results of this code search:

https://github.com/search?q=repo%3AUltimaker%2FCura+%22G28+X0+%3BY0%22&type=code

and check the full end code in the discovered profiles.

### Actual results

That end code is as follows:

```
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 S1318 P266
```

What that does is (apart from some extrusion shenanigans), turn off the steppers, switch to *relative* mode, home *only the X axis* (contrary to what the comment says, `Y0` is commented out!), try to move the head *relatively* in positive Y direction by 180mm (likely out of bed bounds, hopefully the end stops do their job), disengage the steppers again, switch back to absolute mode.

### Expected results

The end code should be doing that "bed presenting" move in *absolute* mode, or alternatively make sure to home Y first. And disengaging steppers only to engage them again right away is also a bit on the non-nonsensical side of things ;)

I'd propose something like would improve things

```
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90 ;absolute positioning
M300 S1318 P266
```

The way the end code currently shipped is formulated, only the firmware/the printer's endstops are preventing harm to hardware, and I've gotten reports by some confused users already in OctoPrint/OctoPrint#4832 that claims that OctoPrint's sanity check of travel moves in files to be printed is malfunctioning, when it is in fact just detecting this weirdness correctly.

### Add your .zip and screenshots here ⬇️

I don't have any logs to share here as I'm reporting an issue in Cura's shipped profile data, please just check your own source as linked above. 

But since I know how it is to have to triage tickets, here's a picture of a cute puppy that I hope brightens your day!

[image]

[picture source](https://unsplash.com/photos/XeOO8ir_YHs)

GregValiant, 2023-08-30:
Thanks for the report.
I don't understand the problem.  That golden retriever puppy printed so perfect it almost looks alive (although it should have a frisbee in it's mouth).

The definition files are submitted to UltiMaker by either the manufacturer or by a member of the community.  In the Vyper definition file is: 
        "author": "ThatGuyZim",
        "manufacturer": "Anycubic",
I didn't check the others.  Problems within a definition file (other than for UltiMaker printers) just aren't UM Cura problems so I will remove the bug label.

This is the Ending Gcode from the Vyper definition file:

M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors
M107
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
G1 Y180 F2000
M84 ;steppers off
G90
M300 S1318 P266

- The first "M84" line should be removed because the following movements are done after the steppers have lost their positioning.  In addition, disabling  the Z stepper with the nozzle over the print is not a good idea.
- "G1 Z+0.5 E-5 ;X-20 Y-20 F{speed_travel} ;move Z up..." has a semi-colon between E-5 and X-20.  I'm not sure what to make of that but it negates the X and Y movement.
- In the same line (although it is effectively commented out) - the {speed_travel} is in mm/sec and the gcode is in mm/min so the travel speed would have been 1/60 of what would be expected.
- G28 X0 Y0 will home the X and Y (the steppers have been disabled at this point) prior to the following movement line...
- G1 Y180 F2000 does not require the G90 because the X and Y just auto-homed so although it is still in "relative" it will work. 
 I would put it in as "G0 F6000 X0 Y{machine_depth}"

I agree that there are issues in the AnyCubic Ending Gcode (for at least the Vyper) but this isn't a Cura problem.  I don't own one of those printers and I don't believe UltiMaker does either.  This will require someone to go through all the AnyCubic definition files and fix them and submit pull requests for the proposed changes.
I'll leave this open so people will see it.  Maybe someone will make the effort and go through all those files.

SimonIT, 2024-10-27:
I stumbled upon this issue because I just installed klipper on my Anycubic i3 Mega and klipper reports an error *Must home axis first* because of this (Klipper3d/klipper#1404).

So if I understand you correctly, I just need to edit this line to remove the first _M84_?
https://github.com/Ultimaker/Cura/blob/a1029d8fe312120c26c271f17ed82a09a1a7144c/resources/definitions/anycubic_i3_mega.def.json#L22

GregValiant, 2024-10-28:
As I mentioned above, there are a lot of mistakes in both the StartUp and Ending Gcodes of Anycubic printers.

Here are my suggestions for both the StartUp and Ending Gcodes for the i3 Mega.  This fixes the speeds, formats the text to Cura style, eliminates "homing" at the end, resets the feed and flow rates (in case you happened to tune them during the print).
The fixes need to be done.  The way they function is simply my preference and is not a "you must do it this way" thing.

`"machine_end_gcode": { "default_value": ";.............Ending Gcode\nM104 S0       ; Turn off extruder\nM140 S0       ; Turn off bed\nM107          ; Turn off layer fan\nG91           ; Relative positioning\nM83           ; Relative Extrusion\nG1 F{retraction_retract_speed*60} E-{retraction_amount} ; Retract\nG0 F{speed_z_hop*60} Z2.0 ; Move up\nG0 F{speed_travel*60} X-5 Y-5 Z2.0 ; Move Z and wipe out\nG90           ; Absolute positioning\nM82           ; Absolute Extrusion\nG1 F{speed_travel*60} Y{machine_depth} ; Present Print\nM84 X Y E     ; Steppers off except Z\nM220 S100     ; Reset Feedrate\nM221 S100     ; Reset Flowrate\nM300 P300 S4000 ; Beep\n;..............End of Ending" },`

There must be a comma at the end if it is not the last override (Python rule)

I don't know if you use the StartUp Gcode.  It also has errors.  This is my suggestion for the StartUp.

 `"machine_start_gcode": { "default_value": ";...........StartUp Gcode\nG21 ;metric values\nG90 ;absolute positioning\nM82 ;set extruder to absolute mode\nM107 ;start with the fan off\nM220 S100 ; Start with Feedrate at 100%\nM221 S100 ; Start with Flowrate at 100%\nG28 ; Auto-Home XYZ\nG1 Z15.0 F{speed_z_hop*60} ;move the platform down 15mm\nG92 E0 ;zero the extruded length\nG1 F{retraction_prime_speed*60} E3 ;extrude 3mm of feed stock\nG92 E0 ;zero the extruded length again\nG1 F{speed_print*60} ; Set the print speed\nG0 Y20 F{speed_travel*60} ; Move to Y20\nM117 Printing...\n;............End of StartUp" },`

These are just in a readable form (without the 'newline' characters).  They could be copied and pasted into the Machine Settings in Cura (rather than the definition file).
`;...........StartUp Gcode`
`G21 ;metric values`
`G90 ;absolute positioning`
`M82 ;set extruder to absolute mode`
`M107 ;start with the fan off`
`M220 S100 ; Start with Feedrate at 100%`
`M221 S100 ; Start with Flowrate at 100%`
`G28 ; Auto-Home XYZ`
`G1 Z15.0 F{speed_z_hop*60} ;move the platform down 15mm`
`G92 E0 ;zero the extruded length`
`G1 F{retraction_prime_speed*60} E3 ;extrude 3mm of feed stock`
`G92 E0 ;zero the extruded length again`
`G1 F{speed_print*60} ; Set the print speed`
`G0 F{speed_travel*60} Y20 ; Move to Y20`
`M117 Printing...`
`;............End of StartUp`

`;.............Ending Gcode`
`M104 S0       ; Turn off extruder`
`M140 S0       ; Turn off bed`
`M107          ; Turn off layer fan`
`G91           ; Relative positioning`
`M83           ; Relative Extrusion`
`G1 F{retraction_retract_speed*60} E-{retraction_amount} ; Retract`
`G0 F{speed_z_hop*60} Z2.0 ; Move up`
`G0 F{speed_travel*60} X-5 Y-5 Z2.0 ; Move Z and wipe out`
`G90           ; Absolute positioning`
`M82           ; Absolute Extrusion`
`G1 F{speed_travel*60} Y{machine_depth} ; Present Print`
`M84 X Y E     ; Steppers off except Z`
`M220 S100     ; Reset Feedrate`
`M221 S100     ; Reset Flowrate`
`M300 P300 S4000 ; Beep`
`;..............End of Ending`

SimonIT, 2024-10-28:
Thanks for the detailed answer! Is there anything against changing it in the repository? I don't mind making a PR for it

GregValiant, 2024-10-28:
Cura 5.9 has 15 Anycubic printers and they really all need to be checked.  Since there haven't been a lot of complaints I figure either nobody cares, or they are fixing the problem locally.
Those snippets I provided should be pretty generic.  They work off the Cura settings rather than having hard coded moves.  The fact that there aren't any hard-coded purge lines helps.

Cora 5.9beta has some problems that need to be resolved so I don't think it's reasonable to assume that any definition changes will make it into 5.9stable.  5.9 is introducing more "logic" into the StartUp and Ending gcodes.  That makes for a lot more flexibility.
I check a lot of gcode and project files.  This is from my own startup for my Ender.

;--------------------------StartUp Gcode
{if 'TPU' in material_type}
M92 E119 ; Steps for TPU
{elif 'PETG' in material_type}
M92 E102 ; Steps for PETG
{else}
M92 E97 ; Steps for PLA
{endif}
{if machine_center_is_zero}
M206 X-116 Y-121 Z0 ; Home offsets when 'OriginAtCenter' is True
{else}
M206 X-1 Y-6 Z0 ; Home offsets when 'OriginAtCenter' is False
{endif}

So the StartUp and Ending can be a lot more personalized.  The printer definitions should have good base codes that work and Anycubic falls short of that.

GregValiant, 2026-09-10:
Puppy or not - I'll close this.

closed 2026-09-10 by GregValiant