Ultimaker/Cura #16622: Strange behavior when using CuraSceneNode.getWorldTransformation() (for export of node properties)

Ant0ine64, 2023-08-30:
### Cura Version

5.4.0

### Operating System

Linux 6.4.12-arch1-1 x86_64 GNU/Linux

### Printer

-

### Reproduction steps

I'm fiddling around creating a plugin, at some point I want to export a scene node (as an stl file) to do some processing.
I didn't (achieve to?) find a method in Cura or Uranium code that could allow me to export a SceneNode, so what i tried is:

- get the CuraSceneNode (I'll call it `node`)
- use `trimesh.load(node.getMeshData()._file_name)` to have it as a trimesh object
- use `trans_matrix = node.getWorldTransformation().getData()` to get the transformation matrix
- apply the transformation matrix (using `mesh.apply_transform(trans_matrix)`

### Actual results

Then I found out that the transformation matrix contain inverted **y** and **z** values, so I inverted theses row and columns.

But in addition some of them are minus values, and sometimes I have to transpose this matrix.

### Expected results

So what's the deal with this transformation, how could I use it.

Or even better, is there a better way to export a node while conserving it's orientation ?

### Add your .zip and screenshots here ⬇️

If i just invert  y and z rows and columns I get something like this:

[image]
The matrix of the left one have been applied on the right

Also, does this concerns Cura or Uranuim ?

5axes, 2023-08-30:
Did you have a look to LocalFileOutputDevice ?

or from UM.FileHandler.WriteFileJob import WriteFileJob

Ant0ine64, 2023-08-30:
@5axes Thanks for the answer, that help me find Uranium/plugins/FileHandlers/STLWriter/STLWriter.py, it's seems to be doing what I want. But it's a plugin and i can't find where (and if) it's used and i can't import it a normal module.
I tried  `PluginRegistry.getInstance().getActivePlugins()` but I can't seem to see it (and nothing in disabled), could this plugin be imported by PluginRegistry ?

Also, the writing with mesh.export() would work if I can handle properly the transformation matrix. But because I can't, using this does not work.

Ant0ine64, 2023-08-30:
Never mind, I avchieve to do what I want with this code:

```python
pr = PluginRegistry.getInstance()

active_pl = pr.getActivePlugins()
Logger.log('d', f"{active_pl=}")

stl_writer = pr.getPluginObject("STLWriter")

with open("/home/antoine/written_piece.stl", 'w') as f:
            stl_writer.write(f, [piece])
```