Ultimaker/Cura #16623: Flsun Sr purge line retracting instead of extruding

GR34VES, 2023-08-30:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Flsun Sr

### Reproduction steps

1. sliced stl 
2. upload to printer via mainsail
3. when print starts with purge line it retracts at a high speed (150mms) throughout the whole line.
4. print still continues as normal but no filament in hotend due to the purge line 

### Actual results

purge line retracts instead of extrudes 

### Expected results

the same situation 

### Add your .zip and screenshots here ⬇️

[FSR_C.zip](https://github.com/Ultimaker/Cura/files/12476061/FSR_C.zip)

GregValiant, 2023-08-30:
Thanks for the report.

This doesn't appear to be a Cura problem and your StartUp Gcode doesn't call for any large retraction.

G28 ; home axis
BED_MESH_PROFILE LOAD=default ;>>>>>>>**This is not gcode so must be for Mainsail??**
**; Lower nozzle** and move to start position
G1 Z150 ;>>>>>>>>>>>>>>>>>>>>>??? **why is this 150mm Z move in here?**  It doesn't really look like "Lower Nozzle"
G1 X-130 Y0 Z0.4 F3000
; Extrude about 40 mm by printing a 90 degree arc
G3 X0 Y-130 I130 Z0.3 **E40** F2700
So there is no long retraction _BUT_ there is also no "G92 E0" prior to that purge extrusion.  You can add one just above that line and maybe it will fix the problem.
The other thing to check is Mainsail and whether it might be configured to add a long retraction.

I'll remove the bug label from this but I'll leave it open for a bit so you can respond here.  I hope you understand.

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant