Ultimaker/Cura #16629: Cannot Slice Lithophane after setting infill to 100%

bacon4bfast, 2023-08-30:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality Ender 3 

### Name abnormal settings

Changing infil to 100% seems to make the bottom layers `999999` and top layers `0`

### Describe model location

In the middle, orientation does not matter.

### Describe your model

Lithophane lamp. Basically a pipe. 

### Add your .zip here ⬇️

Seems that when I change infill to 100% from 20% it makes the bottom layers `999999`. When I try to slice with that, slicing fails. When I change the bottom layers back to 4, I am able to successfully slice the print. 

[CE3E3V2_ItsLitho.zip](https://github.com/Ultimaker/Cura/files/12479912/CE3E3V2_ItsLitho.zip)

bacon4bfast, 2023-08-30:
Behaves similar to  #16464

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic