Ultimaker/Cura #16630: freezes when launching on second monitor

jayandthegames, 2023-08-31:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

ender3

### Reproduction steps

1. Open cura
2. Move to a second monitor
3. Maximize the window
4. Close cura
5. Reopen cura
6. Stuck on a white screen

### Actual results

[image]

### Expected results

load
[image]

### Add your .zip and screenshots here ⬇️

?

lin-ycv, 2023-09-13:
Can confirm this, cura crashed on my non-primary display, and wasn't able to launch since it'd keep launching and get stuck on the display.
Had to unplug all my monitors to have cura launch on the primary display.

shaman1641q, 2023-10-12:
Подтверждаю. Есть экран ноутбука и большой монитор. Если отключить монитор и вывести куру на экране, а потом подключить монитор и  использовать её на мониторе, то она также зависает 
[image]