Ultimaker/Cura #16634: unknown slicer failiure only slices part of the model ; scratched my bed to heck and back when i tried to print

CasualEngineeringYT, 2023-09-01:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

creality ender 3 pro (2023)

### Reproduction steps

1: open ultimaker cura 5.4 ender 3 pro
2: load any model in
3: attempt to slice
4: make sure it isnt a bed placement error
5: make sure item snapping is off
6: try to slice the model supported above the build plate
7: try to figure out why placement and scaling isn't working
(optional): print with this obviously faulty slice hoping it was a mistake and instead have $100 in damage from bed scratches to a thermal runaway (unplugged immediately ender 3)

### Actual results

really bad costly damage up to $100

glad i had extra steppers and another build plate.

### Expected results

what should happen is you get a working slice and gcode

### Add your .zip and screenshots here ⬇️

[image]

[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/12491760/New.Compressed.zipped.Folder.zip)

GregValiant, 2023-09-01:
Thanks for the report and the files.
I went over the gcode you posted and there is a G28 Auto-Home command in your startup but it is immediately followed by Z2.0 to get the nozzle off the build plate prior to any other movement.  There are no Z moves below the 0.2mm of the initial layer.
I sliced  your project file and my gcode did not contain any moves below 0.2mm either.
If you open the gcode in a text editor and search for " Z" (a space followed by "Z") all instances of the z moves will come up.  You can follow the heights to check.  I have never seen Cura enter a "Z0" or a negative Z location and I cannot get my printer to move to Z<0.

This problem is certainly with the printer.  Consider that if this was a Cura bug it would happen to a lot of people (including me and my Ender 3 Pro) and no one would use Cura.  In the 3 years I have been doing triage here this has not come up.

You might have a faulty Z end stop switch (which certainly have been known to fail).  On my printer older E3Pro the Z switch is the same as the Y switch and it's an easy swap.
Make sure you don't have any loose parts on the printer.  A loose hot end can vary position and a loose X beam can make for a variety of problems.

I'm going to remove the bug label from this and go ahead and close it as the gcode you provided is absolutely correct and there is no "bed crash" in it.  I hope you understand.  You can still respond here if you feel this is premature or have additional information on how this could be a problem with Cura.

closed 2023-09-01 by GregValiant

CasualEngineeringYT, 2023-09-05:
thank you for your help and input ive tried the steps on the newest version of cura and it hasnot worked ; that said rotating the print seems to fix the issue for some reason