Ultimaker/Cura #16635: [5.4.0] Slicing with conical supports regardless of if the setting is enabled or not

Toastywf, 2023-09-01:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Sovol SV06 Plus

### Reproduction steps

1. Load model with large [90 degree overhang](https://cdn.discordapp.com/attachments/483268571196162049/1147025819227656262/image.png) 
2. Configure temperatures and supports with conical support disabled
3. Slice using other settings at default
4. Under preview conical supports will be enabled regardless of whether its actually enabled
[image]

### Actual results

Conical supports under Experimental enables itself even if the enable box is not selected, causes unprintable supports resulting in heavy sagging on overhangs as the support below it does not print properly.
[image]

### Expected results

No conical supports as the setting is disabled by default. Fixed by enabling conical support angle and setting it to 0, this should not be necessary as it is disabled in the first place.
[image]

### Add your .zip and screenshots here ⬇️

[Conical bug CURA 5.4.0.zip](https://github.com/Ultimaker/Cura/files/12494309/Conical.bug.CURA.5.4.0.zip)

Asterchades, 2023-09-01:
All third party printer definitions (any which do not list "Ultimaker" as the author) are maintained by the community, as the Cura team lacks the appropriate hardware (and time) to properly test them. As such it's extremely rare that the Cura team will step in to make changes to them.

This particular definition sets the Conical Support Angle to the same 70° it sets the Support Overhang Angle, which suggests whoever made it feels confident this printer (at least as they have it configured) is capable of doing such overhangs without relevant complications. If you disagree with this, and feel the community at large would benefit from having a different default value you are welcome - and indeed encouraged - to submit a pull request altering the default.

The definition in question can be found here: https://github.com/Ultimaker/Cura/blob/main/resources/definitions/sovol_sv06_plus.def.json

closed 2023-09-01 by Asterchades

fieldOfView, 2023-09-01:
Note that the "Setting  Visibility" dialog in the preferences just manages whether a setting is shown or not. It does not affect the value of settings at all. A setting that is not shown to the user still has a value and still affects slicing.