Ultimaker/Cura #16650: Anycubic Kobra 2 slicing problem

MatMal007, 2023-09-03:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic Kobra 2

### Name abnormal settings

none

### Describe model location

no

### Describe your model

the model is only slicable at a very small size

### Add your .zip here ⬇️

[Bug Report Model.zip](https://github.com/Ultimaker/Cura/files/12505680/Bug.Report.Model.zip)

MatMal007, 2023-09-03:
[Repaired model.zip](https://github.com/Ultimaker/Cura/files/12505745/Repaired.model.zip)
i repaired the model and it still doesnt works for unknown reason... please help

GregValiant, 2023-09-03:
Thanks for the report and the file.
It sliced with the layer height at .12 but not at .10.  I'll markk this as a duplicate of a slicing bug that has been reported numerous times.  I'm going to go ahead and close this as the Cura team has numerous examples of this behavior.  I hope you understand.

BTW this won't print very well.  You can see here that a lot of areas don't connect because the 0.40 line width won't allow it.  Even dropping down to a 0.2 nozzle and 0.2 line width there are areas missing and portions of the upper structure will be spaghetti.
[image]

closed 2023-09-04 by GregValiant

MatMal007, 2023-09-09:
Dear GregValiant,

Thank you for your support and Taking time for this Report. Small Question before closing this report, can this bug be fixed by repairing the model?

Greetings,

MatMal007

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

[image]<https://user-images.githubusercontent.com/64202104/265282487-fc60c9d4-f37c-42f5-9010-aa7481296d13.png>

GregValiant, 2023-09-09:
Hello @MatMal007 ,
No.  I repair all models as a matter of course before I try to duplicate a bug.  The problem with this model is that it was never designed for FDM printing.  Some features are just too small to slice so there will be gaps in the print.
You can see how that affects the slice by looking at the machine gun barrel in my previous screenshot.  The barrel ends and the spotlight below simply aren't connected to the main portion of the print.
Those "spotlights" on the front will print, but the brackets that mount them to the body will not because they are just too finely detailed.  The headlights will end up as a pile of spaghetti and will stick to the model.

You could use an app that will allow you to either cut some features off (and get rid of the lights and things) or to make things thicker so they would slice.  I use MS 3D Builder for that sort of thing and there are other programs out there (like Blender) that could do the same thing.  It would be a lot of tedious work no matter what you use.

MatMal007, 2023-09-09:
Hi GregValiant,

I used MS 3D Builder to repair the model and I could already print it bigger. The 3D models are ofcourse straight out of a game's files and are actually meant for in game use. Thank you for helping and I will try to make it work (with removing certain parts of the model that might fix it).

Greetings,

MatMal007

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>