Ultimaker/Cura #16654: cura does not go into slice mode

goofyseeker311, 2023-09-03:
### Cura Version

5.4.0

### Operating System

Win11 RTX3060

### Printer

elegoo neptune 3 pro

### Reproduction steps

loading some stl, slicing them, saving etc, then
loading a gcode file in mistake, then deleting the object,
then opening a 3mf file, it shows "save to disk" and not "slice".
restarting the program restores the ability load and slice 3mf files.

### Actual results

the program thinks it has sliced/modified something being in the save to disk.

### Expected results

it should be in slice mode when a new untitled file is loaded to empty scene.

### Add your .zip and screenshots here ⬇️

random usability error.

GregValiant, 2023-09-04:
Thanks for the report.
I open a LOT of project files and gcodes in Cura and I've never experienced this problem but I'm running Windows 10.  Win11 seems to have some odd issues that didn't occur with Win10.
If you can determine a way to force the behavior in Cura and pass that along it would be really helpful.  It's near impossible to find a bug unless the problem can duplicated reliably.

The Cura team will take a look.

GregValiant, 2023-09-04:
Of course it happened.
I opened a gcode file from another report and then when I opened a project file the "Slice" option was not available.  So as it turns out I was able to recreate the problem.

goofyseeker311, 2023-09-04:
Well curated content, no offense inteb and heavy pun intended.

goofyseeker311, 2023-09-05:
also why does "ESC" key scale down the window, ie, same as the middle button in windows right-top, re-size window button