Ultimaker/Cura #16655: Cura 5.4 gcode does second layer on Skirt

LilBub, 2023-09-04:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Ulimaker S5, Modix 120X, CR-10S Pro v2

### Reproduction steps

1. Slice file with Skirt layer
2. Print
3. Note on the second layer, it goes around the skirt
4. 

### Actual results

Skirt shouldn't be getting anything after layer 1

### Expected results

Skirt should not have a second layer

### Add your .zip and screenshots here ⬇️

[CCR10SPRO_20mm_calibration_cube.zip](https://github.com/Ultimaker/Cura/files/12508383/CCR10SPRO_20mm_calibration_cube.zip)
[CCR10SPRO_20mm_calibration_cube PROJ.zip](https://github.com/Ultimaker/Cura/files/12508384/CCR10SPRO_20mm_calibration_cube.PROJ.zip)

GregValiant, 2023-09-04:
Thanks for the report.
That's a new feature that was introduced in 5.4.0.  If you look under "Bed Adhesion" you will see the setting "Skirt Height".  Setting it to 1 will cause it to go back to the way it was.
[image]

I'll go ahead and close this as it is "expected behavior".

BTW the default value for that setting is "3" and I found it annoying to have to change it to "1" all the time.  I added an override to my printer definition file so that the default height is now 1.
	"skirt_height": { "default_value": 1 },

closed 2023-09-04 by GregValiant

LilBub, 2023-09-04:
Ok, that's good information. 

I have Cura set to ALL for the settings and I had to go into the "slider" icon in Build Plate Adhesion to enable (check off) the Skirt Height, since that isn't viewable by default.

That's a rather silly annoyance they added.  Thanks for pointing it out.  I hope they resolve it

GregValiant, 2023-09-04:
Someone had requested  it to make skirts easier to remove from the build plate.  I never had the problem myself.  I don't think it's going anywhere though.  I fixed it locally (as I mentioned above) so I'm OK now.