Ultimaker/Cura #16659: Support distance and support interface aren't changed with support blockers

Triffen, 2023-09-04:
### Cura Version

5.4

### Operating System

Linux, Manjaro with KDE and X

### Printer

Creality Ender-3 v2 with direct drive

### Reproduction steps

1. Add a model with overhangs, enable tree supports and supports interface
2. Add a support blocker that changes support Z distance and/or disables Support interface
3. Slice

### Actual results

Support interface is still clearly visible on the preview, and the Z distance of 1mm I set should be visible too, but it's not.

### Expected results

Support interface on/off and support Z distance should be changed with support blockers, since they are in the list of settings that can be changed.

### Add your .zip and screenshots here ⬇️

[SupportCalibrationTest-.00-0.zip](https://github.com/Ultimaker/Cura/files/12512601/SupportCalibrationTest-.00-0.zip)
[image]
[cura.zip](https://github.com/Ultimaker/Cura/files/12512659/cura.zip)

GregValiant, 2023-09-04:
Thanks for the report.
There is something wrong with the zip file as there is no project file inside.  I don't think I'll need it.

I'm not sure if this is a bug or just something that isn't explained very well.
If you have the "Enable Support Roof" and the "Support Roof Distance" in the Cutting Mesh (as opposed to "Enable Support Interface" and "Support Z distance") then you get the expected behavior.

The block on the right is configured as you have yours, and the block on the left is configured as I said above.
 
[image]

This appears to be OK if you use the "specific" Child settings instead of the "general" Parent settings.
The Cura team will take a look.

Triffen, 2023-09-04:
Hmm, definitely not explained well, or at all.
But I don't see 'Support roof distance' setting, even with all of them enabled. There is support roof line distance, but that's the distance between the lines of the roof.

GregValiant, 2023-09-04:
The button to the right of the Setting Search box is the visibility tool.  Click it and set the visibility to "All".
I have shut some off here for clarity, but you can see the parent commands and the children commands.
[image]

Triffen, 2023-09-04:
They are set to All.
I don't see 'Support Roof Distance' on your screenshot, or in my settings lists.

GregValiant, 2023-09-04:
That was my bad.  It's "Support Top Distance" and is right under "Support Z Distance".

Triffen, 2023-09-05:
Oh, I have that one!
It looks working in the preview, I'll see for sure after a test print.

ThomasRahm, 2023-09-06:
@GregValiant If one disables Support Floor, then disables Support Interface one can get to the situation where in the per-mesh settings, neither enable support interface, nor enable support roof are clickable.

[image]
[EnableSupportInterfaceNotClickable.zip](https://github.com/Ultimaker/Cura/files/12536297/EnableSupportInterfaceNotClickable.zip)

GregValiant, 2023-09-06:
I think you are running into a Parent / Child override situation.  This might be a bug, or it might be poorly documented behavior.
If you enable the main "Generate Support" setting for the entire project and then set the Per Model settings to kill the support for the two outside models you can put together the settings in the Per Model settings that will allow you to configure the support of the center model so it is selectable.
[image]

So it would appear  that the order in which things are enabled/disabled and the order in which the overrides are applied comes into play.

Right now you have support being generated "Touching Build Plate" so there won't be any support in those models to adjust.
You would need to set that to "Everywhere"