Ultimaker/Cura #16671: Standby temperature not used until AFTER first use of the nozzle

omgitsraven, 2023-09-05:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creator Pro 2

### Reproduction steps

1. Make an IDEX print
2. Set a noticeably lower standby temperature for your second nozzle
3. Set the raft to print with your FIRST nozzle
4. Slice, and begin printing.

### Actual results

The second nozzle sits at its PRINTING temperature, while it's sitting waiting several minutes for the first nozzle to finish printing the raft.  (After it has finished printing its first layer, THEN it lowers down to its standby temperature, but by that point it's already been burning the first few mm of filament for several minutes.)

### Expected results

The second nozzle should be at its STANDBY temperature while it's waiting to print for the first time.

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2023-09-06:
> The second nozzle sits at its PRINTING temperature, 

This probably depends on your start gcode snippet. Can you paste your start gcode snippet as it is shown in the Machine Settings dialog?

omgitsraven, 2023-09-06:
Excellent point; you're right, this is probably it.

```
M118 X30.00 Y30.00 Z25.80 T0 T1
M140 S{material_bed_temperature_layer_0} ;start bed preheating
M104 T0 S{material_print_temperature_layer_0, 0}
M104 T1 S{material_print_temperature_layer_0, 1}
M107
G90
G28
M132 X Y Z A B
G1 Z50.000 F420
G161 X Y F3300
M7 T0
M6 T0
M6 T1
M651
```

I got this off some forum post somewhere, because without it, my two print heads crashed into each other...

I'm guessing I need to replace `material_print_temperature_layer_0`, but do you know how I'd determine what the name of the variable for the standby temperature is instead?   (And whether that would be safe to leave in permanently, even when not specifying a standby temperature?)

GregValiant, 2023-09-09:
{material_standby_temperature} and you might find "T{initial_extruder_nr}" useful.

Cura does not directly support IDEX printers.  They can be made to work with Cura, but you have to use workarounds (like in your case).  There is no support for your A B axes.  Some machines use U V so there seems to be no standardization there.

I'm going to remove the bug label and close this.  Let us know if you feel that is premature.
If you remove the temperature lines from the startup gcode and allow Cura to handle it you may get a better result.

closed 2023-09-09 by GregValiant

omgitsraven, 2023-09-09:
I think closing it is fair, thanks.   I'll give that a shot soon.   (I might have a related bug to open soon where I think the standby temperatures aren't totally being respected correctly *either*, which I have a feeling will come up in attempting to test this, but I'll make that its own thread since this immediate problem is definitely an issue with my start gcode; thanks for your advice with it!)

GregValiant, 2023-09-09:
This would be the ticket but unfortunately it doesn't work.
M104 T0 S{material_standby_temperature, 0}
M104 T1 S{material_standby_temperature, 1}
M109 T{initial_extruder_nr} S{{material_print_temperature_layer_0}, {initial_extruder_nr}}
Cura doesn't assess or actually read a replacement, it just sticks in the number and as a consequence that M109 line throws an error because it would need to be "evaluated".
I've been playing with a post processor for the Raise3D IDEX printers, but they are simpler and use an M605 code to pass the "Normal", "Mirror", "Duplicate" commands to the printer.  I did notice that Cura is real good at starting a print with the correct extruder and that's why I advised to leave the temperature lines out of the StartUp and just let Cura handle it.  What you really need is a post processor that would replace a dummy line in the startup with a real temperature line for the initial extruder.

GregValiant, 2023-09-09:
This is from the FlashPrint Cura installation manual for the Creator Pro2:
_"Note: Cura sliced files are only available for printing with Creator Pro 2’s right extruder."_
I think we can assume the reason for that is the lack of Cura support for the A and B axes of the left extruder.
On the other hand - now you don't have to worry about it.

Wiriza01, 2025-03-03:
In general came across this post as I was looking to find a Cura setting for the definition of how long it takes to bring temp back up from standby (lets call it Txs).  was hoping that Cura would calculate set T2/T1 based off the Tx print time (Time it takes to complete print layer for that nozzle - Txs.  
Example.. T1 finished and parked w/ standby.. T2 is up now and has a long job.. 2min.   T1 should be ready to print say 2min-4Seconds where at that time period gcode line to warmup T1. as to not have to wait for T1.   T2 could have been a short job as well, perhaps there could be a value of no standby temp IF alt nozzle job is at least x value.  If there is already place I am looking in the wrong places, but hope this form is a start....

GregValiant, 2025-03-03:
With the "Printer Settings" plugin installed from the MarketPlace there are two relevant settings available:  "Heat up Speed" and "Cool down Speed".  The defaults for both are 2°/second.  They are "settable per extruder".

Cura will try to make adjustments for short layer times.  If you want to drop from 240 to a standby of 170 then at 2°/sec it would take the nozzle 35 seconds to cool down, and 35 seconds to heat back up.  If M109 is used then another 10 seconds can be added because it will wait for the temperature to settle down.  That would mean a minimum layer time of 80 seconds.  Cura will try to make adjustments, but it is imprecise.

Wiriza01, 2025-03-03:
Downloading as I type this response.. Thanks.. kept going back to the Dual Extrusion options in cura.. Fingers crossed but w/ that said anything for quicker prints right!?   Thanks again.

GregValiant, 2025-03-03:
As I recall, Cura will adjust the standby temperature upwards when a layer is too short (timewise) to allow the nozzle to drop all the way to standby.  The 10 second hit caused by M109 will still exist.  Nothing to do about that except to change them all to M104's but then you can start with an extruder that isn't up to printing temperature and that usually isn't acceptable.

A recent addition to Cura was addition of logic capability to the StartUp Gcode.  Now you can do something like this:
```
; Start G-code
{if initial_extruder_nr == 0}
M104 S{material_standby_temperature, 1} T1 ; Warm up second extruder
M109 S{material_print_temperature_layer_0, 0} T0 ; Heat initial extruder
{else}
M104 S{material_standby_temperature, 0} T0 ; Warm up first extruder
M109 S{material_print_temperature_layer_0, 1} T1 ; Heat initial extruder
{endif}
M190 S{material_bed_temperature_layer_0}
```

That will heat whichever nozzle will be used first, while leaving the unused nozzle at Standby Temp.  How that will be impact what goes on at the first tool change you will have to check.