Ultimaker/Cura #16676: slicing failed with two different stl on build plate

sasillick, 2023-09-06:
### Cura Version

5.4.0

### Operating System

MacOS Monterey

### Printer

Ender 3 v2 Neo

### Describe model location

fails wherever models are

### Describe your model

router copy inserts - print OK singularly

### Add your .zip here ⬇️

[CE3V2Neo_bagues_de_copiage_diam_15.3mf.zip](https://github.com/Ultimaker/Cura/files/12535261/CE3V2Neo_bagues_de_copiage_diam_15.3mf.zip)

sasillick, 2023-09-06:
After further testing, if supports are turned off then Cura slices OK

GregValiant, 2023-09-06:
Thanks for the report and the files.
I didn't have any trouble slicing it on Windows.  They are pretty simple models and although there is a minor error someplace I think they should have sliced.
The Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic