Ultimaker/Cura #16678: Almost identical Profiles producing vastly different first few layers.  One is broken badly.

proxyissues, 2023-09-06:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Voron2,  stock firmware

### Reproduction steps

I've included 2 saved projects.  The PLA named file that shows slicing problem cannot slice some models as well.

Bottom layers are wrong and have 1mm gaps.   I have modified the PLA profile to use same temperature and settings to the ABS in an effort to find WHAT setting is breaking the slicing so badly.

### Actual results

The broken project has wall gaps exceeding 1mm on first layer and few first layers that do not exist using the working profile.  The settings are the same from visible fields.

### Expected results

Slicing lines should be identical between two profiles when using same settings (line width, cooling, line height, etc...)

### Add your .zip and screenshots here ⬇️

[image]
[Broken-and-working.zip](https://github.com/Ultimaker/Cura/files/12538932/Broken-and-working.zip)

GregValiant, 2023-09-06:
Thanks for the report.
There are errors in the model.  At least one face is inverted and that's what is causing the problem as the model "is not water-tight and may not print correctly".  You should load the "Mesh Tools" plugin from the Marketplace so you get the warning and when you do get the warning it should not be ignored.

Here is your original model on the right and on the left is the model repaired by MS 3D Builder.
[image]

As to why one profile sliced it as expected, that's one of those "inconsistencies" that show up when models are broken and that's why the warning exists.
This report is from formware.co/OnLineSTLRepair.
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **130 Non-manifold edges** (?)
--> **203 Inverted faces** (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 3996ms ------
-> Vertex count changed from 930 to 3191 (+2261)
-> Triangle count changed from 2220 to 6374 (+4154)

I'll remove the bug label from this but I'll leave it open for a while so you can respond.  I hope you understand.

closed 2023-09-06 by proxyissues

GregValiant, 2023-09-06:
It is actually pretty rare for Fusion 360 to produce a bad model.  You need to be careful when merging/subtracting/intersecting.  Almost all the errors I've seen are from users not properly constraining their sketches or not merging sub-models correctly and leaving gaps.