Ultimaker/Cura #16679: JP-Orange Pi 3 LTS Chassis

rsc2023, 2023-09-06:
### Cura Version

5.4.0

### Operating System

Windows 11 Pro

### Printer

Sovol 6 Plus

### Name abnormal settings

changed supports from normal to tree but that is all

### Describe model location

back center, front left corner and front right corner (x-68)(Y-108.6)(Z-25)

### Describe your model

It is 3 pieces to form a chassis for an Orange Pi 3 LTS. First time printing this model. Downloaded the design from Thingiverse.com

### Add your .zip here ⬇️

The original design had a solid bottom. I thought it would take too much time to print properly so I changed it to the lattice. 

[Orange_Pi_3.zip](https://github.com/Ultimaker/Cura/files/12541098/Orange_Pi_3.zip)

GregValiant, 2023-09-07:
Thanks for the report and the file.
I think the grid pattern you added needs a re-do.  My normal tools aren't showing any errors but here is the slice in Cura.  You can see that surfaces aren't flat and I'm not sure what's going on with that top surface being added across that hole.
[image]

This view is in MS 3D Builder.  These areas should be flat but they are all spiky and the beams that make up the grid aren't even.   
 I tried to modify the model but I couldn't alter it and that is odd.
The model seems to be watertight, but there is something going on that Cura doesn't like.
[image]

I'll leave the bug label on this because I can't really point to a specific thing, but this sure looks like a problem with the model.

rsc2023, 2023-09-07:
Hello,

Thank you for getting back to me, that doesnt happen very often. The original file had a solid bottom where the lattice is, but io changed it because I was afraid it would become too hot when I put the SBC where it will be going, which is in the very bottom of my PC chasis and will control an LCD Display mounted inside of my PC tower as a cool video animation style effect. I have no clue about the top piece, but ate last night I took it off of the file and left the other 2 pieces and it sliced just fine.

Question, do you know of better software I could use other tha 3D Builder? 3D Builder works, but is very minimal. I looked at few other programs as well, but would like a bit of advice on it.

Thank You,
Jason Page

GregValiant, 2023-09-07:
Blender has fans.  It has a bit of a learning curve.
I use 3D Builder for altering STL files.  It isn't very intuitive but it does a good job and "generally" leaves clean cut results.
I don't know what the original model looked like but if you bring the original model into 3D Builder and just start subtracting cubes from the bottom of the original model you should be good to go.  I know...locating things in Builder isn't easy, but it should get the job done.
As far as designing  your won stuff Fusion 360 may be the best of the lot.  Sketchup and ones like it just aren't very good.  You can get a model built, but it will have issues unless you are very, very careful.
No matter what you choose there is going to be a learning curve.  But the skills will come in handy for all sorts of things.

rsc2023, 2023-09-07:
Thank you.

Thank You,
Jason Page

-------- Original Message --------

GregValiant, 2023-09-08:
@rsc2023 I'm going to go ahead and close this.  The model did slice for me but it has serious issues that need to be addressed.  I hope you understand..
Good luck with it.

BTW I printed a little circuit board box for power relays the bed and hot end on my Ender.  I put stand-offs on the bottom to keep the boards from contacting the bottom and put vents in the top above the heat exchanger fins so the top of the box wouldn't warp.  That seems to work well.
Any heat is going to want to go up.  As long as the board isn't in contact with the plastic and has an air gap below it and room on the sides for any warm air to escape you should be good.  Look at it this way - you can always revise the design and print another one.

closed 2023-09-08 by GregValiant