Ultimaker/Cura #16687: Slice failed

JoshIsEpic, 2023-09-07:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Ender 3 V2

### Name abnormal settings

M420 S1

### Describe model location

90 degrees

### Describe your model

Ninja trophy for CodeNinjas

### Add your .zip here ⬇️

[CE3E3V2_NinjaTrophyFixedV2.zip](https://github.com/Ultimaker/Cura/files/12553234/CE3E3V2_NinjaTrophyFixedV2.zip)

5axes, 2023-09-13:
Windows 10 Cannot reproduce
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic