Ultimaker/Cura #16688: Slicing failed

mikemsd, 2023-09-07:
### Cura Version

5.4.0

### Operating System

MacOS Ventura on M1 MacBook

### Printer

Ankermake M5 CE

### Describe your model

Slices successfully in AnkerMake Slicer. Yes tried fix with Meshtools with no success

### Add your .zip here ⬇️

[AM5CE_AtlantaLion.3mf.zip](https://github.com/Ultimaker/Cura/files/12554029/AM5CE_AtlantaLion.3mf.zip)

5axes, 2023-09-13:
ankermake_m5_ce  printer is not in the standart list of the Cura Printer. Cannot reproduce the issue without this element.  Works fine with a standard printer. 

[image]

5axes, 2023-09-13:
An other origin ... This model is made with Three mesh 
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic