Ultimaker/Cura #16698: Infill multiplicator

clay473874, 2023-09-08:
### Cura Version

5.3.0

### Operating System

Windows 10

### Printer

Anycubic Kobra Go

### Reproduction steps

If you set a multiplier for the grid filling and then go to zigzag, the multiplier can no longer be removed unless you switch back to the grid filling pattern and change it there again

### Actual results

infill multiplier can not be changed in zigzag

### Expected results

change multiplier in zigzag mode

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

GregValiant, 2023-09-08:
Thanks for the report.
When the new controls were introduced there were some tricks missed.  This looks like another one.  The setting is (currently) supposed to be hidden when ZigZag is selected, but then it shouldn't be used either.  If it is going to be in play then I agree the setting should be visible.

"enabled": "infill_sparse_density > 0 and infill_pattern != 'zigzag' and (gradual_infill_steps == 0 or not zig_zaggify_infill)"
So it is enabled if: the Infill Density is >0 and the Infill Pattern is anything but "zig zag" and Gradual Infill Steps = 0 or Connect Infill Lines is False.

@MariMakes and the Cura team will take a look.

MariMakes, 2023-09-11:
Hi @clay473874,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

That definitely looks like a Bug on our side. 🐛 
[image]

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix it will not get a high priority to implement.
Fingers crossed 🤞

MariMakes, 2023-10-20:
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11199

Thanks for the report! 👍