Ultimaker/Cura #16700: Speicherung zum Drucken

effjott, 2023-09-08:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 2

### Reproduction steps

Hallo,
Keine Endung "gcode" zum speichern z.B. auf Festplatte, nur "stl und obj"
MfG FJHomberg

### Actual results

Hallo,
Keine Endung "gcode" zum speichern z.B. auf Festplatte, nur "stl und obj"
MfG FJHomberg

### Expected results

Hallo,
Keine Endung "gcode" zum speichern z.B. auf Festplatte, nur "stl und obj"
MfG FJHomberg

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-09-15:
That looks like the "Export" dialog.  After slicing a model the slice button will change to two buttons "Preview" and "Save to Disk". 
 Use that "Save to Disk" button (far lower right) to save your gcode.
If that worked for you let me know (let me know if it didn't work as well).  Right now it doesn't look like a bug.
[image]

GregValiant, 2023-09-19:
Images do not come through via email.  They must be inserted here on the website.

When you go to "Save to Disk" there is no option for " ***.gcode**" and there should be.

@MariMakes of the Cura team will take a look at this.  Whether you use the "Save to Disk" button or the "Export" command there should be an option for "**G-code File (*.gcode)**" but there may be situations where it is not an option, or it could be a bug.

effjott, 2023-09-19:
Ok, I'll check it out.

fieldOfView, 2023-09-19:
@effjott posted this elsewhere:

> Dear Greg Valiant
> What I mean is the ending .gcode, which is not intended for file storage, only .stl and .obj if I want to save to hard drive, for example.
> Regards FJHomberg

[image]

closed 2023-11-02 by github-actions[bot]